mvnmgrx/kiutils #31: KiCad 7 feature collection

mvnmgrx, 2022-09-20:
A collection of new stuff KiCad 7 seems to implement that may be important in the future.

[April developer update:](https://www.kicad.org/blog/2022/04/Development-Highlight-April-Update/)
- Symbol: New pin types, such as Unit, Orientation, etc
- Board: Inverse text objects

[February developer update:](https://www.kicad.org/blog/2022/02/Development-Highlight-February-Update/)
- DRC: Mechanical clearance rules: Two new constraints (`mechanical_clearance` and `mechanical_hole_clearance`)
- DRC: Custom rule severities (new `severity` token added!)
- DRC: Pad to zone rules (`thermal_relief_gap`, `thermal_spoke_width` and `zone_connection`)
- Dimension: Radial dimension tool
- Schematic: Rectangle and circle primitives added
- Schematic: Custom fonts in development
- Schematic: Text boxes added

hlprasu, 2023-02-19:
The stroke definition seems to have made the color property to be optional in the KiCad 7.0.0. Unfortunately this is undocumented but visible in generated .kicad_sym files. 

Specifically, in https://dev-docs.kicad.org/en/file-formats/sexpr-intro/index.html#_symbols, the stroke definition given is 
```
  (stroke
    (width WIDTH)                                               
    (type TYPE)                                                 
    (color R G B A)                                             
  )

```

This is valid for KiCad 6.0. But, in KiCad 7.0 it seems to be 
```
  (stroke
    (width WIDTH)                                               
    (type TYPE)                                                 
    [(color R G B A)]
  )
```
where color property is made optional for a stroke. This needs to be revalidated with KiCad team which I am posting to them for confirmation.

hlprasu, 2023-02-21:
```
diff --git a/src/kiutils/items/common.py b/src/kiutils/items/common.py
index bca15f8..5384ed9 100644
--- a/src/kiutils/items/common.py
+++ b/src/kiutils/items/common.py
@@ -233,7 +233,7 @@ class Stroke():
         Returns:
             - Stroke: Object of the class initialized with the given S-Expression
         """
-        if not isinstance(exp, list) or len(exp) != 4:
+        if not isinstance(exp, list) or len(exp) < 3:
             raise Exception("Expression does not have the correct type")
 
         if exp[0] != 'stroke':
@@ -246,6 +246,9 @@ class Stroke():
             if item[0] == 'width': object.width = item[1]
             if item[0] == 'type':  object.type = item[1]
             if item[0] == 'color': object.color = ColorRGBA.from_sexpr(item)
+        
+        if len(exp) == 3:
+            object.color = ColorRGBA() # 0, 0, 0, 0 -- default
         return object
 
     def to_sexpr(self, indent=2, newline=True) -> str:
```

I did the above change in code which fixed it. Does this seem ok?

mvnmgrx, 2023-03-27:
Closed in preparation of next release.

closed 2023-03-27 by mvnmgrx