mvnmgrx/kiutils #46: Missing "lib_name" field in schematicSymbols

FutureEngineeer, 2023-02-07:
Description: The optional field "lib_name" is missing in schematicSymbols and is used to create alternative symbols directly in the schematic. As a result, components modified in the schematic are not displayed.

Example from schematic file: (symbol (lib_id "") (lib_name "USBLC6-4SC6_1") (at 43.18 135.89 0) (unit 1)

Solution: The field can be added by adding the following code to the class SchematicSymbol() in schitems.py:
``` python
libraryName: str = ""

if item[0] == 'lib_name': object.libraryName = item[1] 
# in from_sexpr

f'(lib_name "{dequote(self.libraryName)}") 
# after f'{indents}(symbol (lib_id "{dequote(self.libraryIdentifier)}")
```

mvnmgrx, 2023-02-13:
Can't find this token in the [KiCad documentation](https://dev-docs.kicad.org/en/file-formats/sexpr-schematic/#_symbol_section) for schematic symbols .. 

What KiCad version your file belongs to? In KiCad 6, the lib name might be insoide of "lib_id", we are currently discussing similar changes for the Library Symbol class in #41 

Can you provide a full example of what you are trying to parse?

FutureEngineeer, 2023-02-13:
When creating an alternative version of a symbol directly in the schema (KiCad 6), it adds an additional prefix _1 or higher to the lib id, depending on the number of variations of the modified library symbols in the schema. If the libname field is missing, then such components are no longer displayed in the schema after saving. After these changes in the code, the problem was solved

mvnmgrx, 2023-02-14:
It wouldn't be the first time that a token is missing in the KiCad documentation. I'm going to take a closer look on this later this week, thanks for sharing the example.

PR #54 (2023-02-20): Feature: Consistent library id APIs for Footprint, Symbol and SchematicSymbol & new lib_name token
This PR implements the following:
#### The library ID API for `Footprint`, `SchematicSymbol` and `Symbol` was consolidated to the following:
 - The `libId` token provides a getter/setter method for combined library IDs like `MyLibrary:MyPart`. It parses the given ID into its parts, depending on the class being used:
   - **Symbol**: `self.libId` may consist of `self.entryName`, `self.libraryNickname`, `self.unitId` or `self.styleId`, depending on the context the symbol is in (top-level symbol or  [... cut]
closed 2023-02-20 by mvnmgrx