mvnmgrx/kiutils #47: Error reading the sch and pcb file (and possibly not only) due to a unicode error

FutureEngineeer, 2023-02-09:
Problem: if there are characters in a file (for example, in a text field) that are not contained in the standard encoding, then a UnicodeDecodeError error occurs: 'charmap' codec can't decode byte: character maps to <undefined>

Solution: when opening a file and writing to a file, you can use encoding='cp437'

For example in schematic.py: 
``` python3
with open(filepath, 'w', encoding='cp437') as outfile:
   outfile.write(self.to_sexpr())
```

mvnmgrx, 2023-02-13:
This may fix your problem, but it might also break the parser for other users .. KiCad as well as the  `open()` function uses UTF-8 as default as far as i know. So it seems like the file you are trying to parse is not ASCII or UTF-8 encoded.

I won't change the default encoding in all file-related functions, but i could add another parameter with which you could specify a different encoding (if converting your files to UTF-8 is not possible).

FutureEngineeer, 2023-02-13:
Thank you, I think this can solve the problem for files in which using languages other than English

mvnmgrx, 2023-02-15:
Added a new parameter `encoding` to all file-related functions in #50. Will be released in v1.2.2 asap later this week. 

Thanks for pointing this out!

PR #50 (2023-02-15): Feat: Added encoding parameter to file-related functions
This PR adds a new optional parameter `encoding` to file-related functions.

Fixes #47
closed 2023-02-15 by mvnmgrx