mvnmgrx/kiutils #48: No ability to set library table entry as inactive

zardini123, 2023-02-15:
KiCad's GUI provides the functionality of setting a symbol or footprint library entry to inactive.  When a symbol or footprint library table entry is set to inactive, the token `(disabled)` is added to the corresponding entry in `sym-lib-table` or `fp-lib-table`.  When the entry is active, there is no token, and therefore is assumed to be active.

Example below shows `symbol_lib_1` as active, and `symbol_lib_2` as inactive:

```
(sym_lib_table
  (lib (name "symbol_lib_1")(type "Legacy")(uri "${KIPRJMOD}/legacy_libs/legacy_lib_1.lib")(options "")(descr ""))
  (lib (name "symbol_lib_2")(type "Legacy")(uri "${KIPRJMOD}/legacy_libs/legacy_lib_2.lib")(options "")(descr "")(disabled))
)
```

As of currently, kiutils does not provide an API or functionality for setting a library in a library table as active or not.  I would suggest adding a member variable `active` to `kiutils.libraries.Library` as to reflect the KiCad naming convention in the symbol and footprint library managers.

Thank you all for kiutils!

zardini123, 2023-02-15:
As my current project is currently directly dependent on the active/inactive functionality, I would be interested in creating a fix and opening a PR!

mvnmgrx, 2023-02-15:
Hello @zardini123 
Thanks for pointing this out. Should be a quick fix, i'm looking forward for a PR on your proposed changes :-) 

Best regards

zardini123, 2023-02-17:
Hi @mvnmgrx , I have wrote a fix, but now I am working on tests but having issues executing you all's current tests.  How do you go about executing tests for `kiutils`?

For me attempting to run `test.py` in root (`python3 test.py`) results in `ModuleNotFoundError: No module named 'kiutils.footprint'` as the unit test runner does not have a reference to the `kiutils` module being under `src/kiutils`.

zardini123, 2023-02-17:
I managed to figure out how to run tests, but required changes to `test.py`.  I have provided those changes in #52.

mvnmgrx, 2023-02-18:
Hi @zardini123 

I usually ran tests by either installing the `kiutils` sources using `pip install -e .` or by adding the sources to the PYTHONPATH environment variable by hand. But i agree that doing that in our `test.py` makes more sense, especially when kiutils is already installed on the system the tests may test the wrong source code when using my approach.

PR #51 (2023-02-18): Added `active` member to kiutils.libraries.Library for ability to disable specific libraries
Added 'active' boolean member to kiutils.libraries.Library.  In addition, modified Library/LibTable tests to include the member.

This member reflects KiCad's own functionality.  When `active == False`, a `(disabled)` token is present in the library's S-expr.  When `active == True`, the token is absent.

Fixes #48.
closed 2023-02-18 by mvnmgrx