mvnmgrx/kiutils #56: API: Do not allow multiple library table entries of same nickname

zardini123, 2023-02-18:
With library tables in KiCad (`sym-lib-table` and `fp-lib-table`), KiCad cannot accept more than one entry with the same nickname (see screenshot below).  With this, it seems KiCad processes the library table as a dictionary where the key is the nickname.

As of currently KiUtil's API for library tables does not organize library table entries by a key.  Instead it is an array of library table entries.  Assuming all users of KiUtils intend to import their symbol tables to be used by KiCad >= 6.0, maybe it would be of use to have libraries be a dictionary, where the key is the nickname?

This change in API would enforce invariants imposed by KiCad.

[image]

zardini123, 2023-02-18:
I would still see use in having the nickname still present in the `kiutils.libraries.Library` object.  Maybe instead of having a duplicate of the key in `kiutils.libraries.LibTable`, the `kiutils.libraries.Library` could be hashable based on nickname?  Then `kiutils.libraries.LibTable.libs` being a set instead of a dictionary would enforce the invariant and not have clones of the key.

zardini123, 2023-02-20:
A similar philosophy of dictionary or set could be useful also for properties in symbols.  Properties have a key, and for quick lookup and approachable API could be done with a key/hash based data structure.

mvnmgrx, 2023-02-20:
The problem seems that for a set to work correctly, its items need to be immutable (or at least the value used to hash and compare the items must be immutable for the whole lifetime of the object, see https://docs.python.org/3/glossary.html#term-hashable). 

So something like this would work, but would be invalid as the `name` property may be changed later on:
```python
@dataclass
class Library():
    # ...
    def __eq__(self, other: Library) -> bool:
        return self.name == other.name

    def __hash__(self) -> int:
        print(f"Hash of {self.name}: {hash(self.name)}")
        return hash(self.name)
```

Making the dataclass object frozen would fix this but it defeats the purpose of the dataclass here, as i would like to be able to change members without reassigning the whole object to something else because it was copied..

A dict-based implementation would work, but then we have to deal with essentially two `name` parameters that must be in sync ... 

We could maybe make `name` a property without a setter method. But this technically does not make property immutable ..

zardini123, 2023-02-22:
> Making the dataclass object frozen would fix this but it defeats the purpose of the dataclass here, as i would like to be able to change members without reassigning the whole object to something else because it was copied..

Ah yeah, I forgot about this part.  It is critical for the objects to be mutable.

Though even with a dictionary, the only way to change a value to a new key is remove the entry from the key first.  Therefore, the `Library` entry would have to be removed before it could be changed to a new name.

Maybe instead should there be a set of API for managing library entries in the `LibTable` class?  For example `add_entry`, `remove_entry` `change_entry` etc?  The array of `Library`s would be private.  This could then enforce specific invariants regardless of the underlying data structure.

mvnmgrx, 2023-06-20:
So i thought some options through and tested this using sets, ordered sets and even a selfmade "unique list" class.

But if we start adding so much API functions here, we would have to add such APIs to almost every class that holds unique items (see properties of schematics, footprints, etc). This adds enormous overhead in developing and testing and is, in my opinion, just not neccessary for what `kiutils` is.

Lets keep it simple and stick with the API-less class structure that we've used everywhere. I will add to the documentation of the ``LibTable`` class that its items need to have unique names and that the application needs to test for this (which is like 2 lines of code to do so). But thats as far as i would go on this topic.

Hope you understand

closed 2023-06-20 by mvnmgrx