mvnmgrx/kiutils #62: Problems with KiCad 7

chfriedrich, 2023-02-23:
Hi,
thank you for the great parser, it helped a lot! I used it to load schematics and symbols generated by kicad 7, manipulate them, and save them again. This is what I noticed:

**Symbols Library**
An exception was thrown when I loaded the .kicad_sym file, because "stroke" has only 3 parameters in kicad 7 instead of 4.
After removing "or len(exp) != 4" from line 236 in common.py the import worked fine.
But there are some differences between the read and the written file:
- In the written file, every property has the parameter (id 0) attatched, like (property "Reference" "R?" (id 0) (at 1.27 0.635 0) instead of (property "Reference" "R?" (at 1.27 0.635 0). But kicad 7 does not complain about that, when I open the file.
- Every stroke has the "color" parameter attached, like (stroke (width 0.254) (type default) (color 0 0 0 0)) instead of (stroke (width 0.254) (type default))

**Schematic**
Reading and writing worked fine, but there also are some differences between read and written file:
- Unfortunatelly the symbols "do not populate" option is not supported yet, so (dnp yes) is missing in the written file
- (id 0) is inserted at any property

Do you have any idea when "do not populate" will be supportet. Unfortunatelly it is very important for my project.

mvnmgrx, 2023-02-27:
Hello @chfriedrich,

Glad you like `kiutils`! I'm currently in the process of switching to KiCad 7 myself and to check what has to be implemented into `kiutils`. As there were some changes that would break compatibility for oder versions, this may take some time..

> (id 0) is inserted at any property 

This is unfortunately one example, here some tokens are made optional which were required in last KiCad version. We're currently discussing on how to deal with this without breaking support for KiCad v6.

The problem with the symbol libraries is currently being fixed in #57. Feel free to check this PR out.

> Do you have any idea when "do not populate" will be supported? Unfortunatelly it is very important for my project.

Little things like this should be added in the next couple weeks, but i can't guarentee you some schedule as i am pretty limited on free time currently.

Best regards

mvnmgrx, 2023-03-27:
Closing this issue, as all items will be implemented in the next release.

closed 2023-03-27 by mvnmgrx