mvnmgrx/kiutils #113: Broken as of KiCAD 8?

evantandersen, 2024-02-29:
It seems KiCAD 8 changed the file format enough that KiUtils opening, parsing, and saving a PCB will corrupt it and prevent it from being opened. 

To replicate:

1. Open a KiCAD 7 PCB in KiCAD 8, you will get a notice about upgrading the file format upon save.
2. Save the PCB to change to new format
3. `Board().from_file("foo.kicad_pcb").to_file()`
4. Try to open it in KiCAD 8, no dice!

mvnmgrx, 2024-04-04:
Hello, KiCad 8 is not yet supported and its possible that there are breaking changes from 7 to 8 which `kiutils` does not yet account for. I haven't yet had time to look into the changes KiCad 8 brought, but its on my agenda to do so. 

For the use-cases you described, the pcbnew internal scripting interface is of course more suited. However, i wrote `kiutils` initially to be able to process KiCad-related files outside of the KiCad environment. A few other use-cases for `kiutils` might include:
- automatic script-based generation of footprints, parts, boards, etc.
- commit hooks in git (an example would be removing filled zones before committing changes into the database)
- parts management outside of KiCad (e.g [kitree](https://github.com/mvnmgrx/kitree) or [Ki-nTree](https://github.com/sparkmicro/Ki-nTree))
- you name it

so, regardless of what happens to the pcbnew scripting interface, `kiutils` will still deliver value in being independent from KICad and lightweight to integrate.

chfriedrich, 2024-06-03:
Hi,
do you have any idea when kiutils will be usable for KiCad 8 files?

samkristoff, 2025-01-21:
Hey @mvnmgrx thanks for creating this project.

I'm building a tool outside of KiCad to do symbol library and symbol manipulation / automation using kiutils.  

I've started updating the to_sexpr() function in the SymbolLib and Symbol classes as well as their dependencies to match the Kicad v8 format (using tabs, updated indenting, etc).  I've just been updating the to_sexpr() directly in a fork of kiutils.  

Do you have thoughts on how best to add v8 support?  Do we want to keep backwards compatability for generating output in older Kicad formats or is updating the to_sexpr() to v8 and dropping legacy output format acceptible?  We could pass the output version as a parameter to the to_sexpr function, but it might be a pain to maintain.

Let me know how you suggest to proceed with this and I'll try to make updates such that they can be merged back in here.

samkristoff, 2025-02-02:
For anyone interested I've updated most of the symbol related stuff to work with KiCad v8 in [this fork/branch](https://github.com/makerhub/kiutils/tree/samkristoff/dev)

My main goal was to be able to edit a symbol library and it's symbols in KiCad v8 and with this tool and be able to git diff the library without a bunch of formatting deltas.  It currently works with my main KiCad symbol lib, but is likely not perfect.

mvnmgrx, 2025-11-25:
Hi @riccardostokker,
thanks for reaching out. I currently have much private stuff that unfortunately keeps me from developing on kiutils at the moment. I haven't even taken a look into KiCad 8 yet KiCad 9 because i had no board projects done for quite a while.

I am open to have other maintainers in the project besides me as quite some improvement merge requests piled up and i have not that much free time at the moment. I'll add you as maintainer if you want to.

Last time i glanced at KiCad 8 S-Expr output i saw that they did quite some changes on formatting. There might be a need for some type of version variable that one might set to adjust in which format the s-expression is generated. It would be nice if we could keep the old v7/v6 styles, as there exist a lot of tests for those formats. 

But we also could talk about kiutils 2.0 if there is no straight forward way to have v6/v7 format and v8/v9 format coexist. I agree that some generic S-Expression formatter would be nicer as it would eliminate the need to generate S-Expr by hand as done currently.

Looking forward for your thoughts on this.

BR,
Marvin