mvnmgrx/kiutils #120: kiutils doesnt save KiCad Schematic properly

brian-geicp, 2024-09-18:
* kiutils-1.4.8.dist-info
* Schematic Generated by 'KiCad Schematic Editor x86_64 on x86_64'
   - Version: 8.0.4-8.0.4-0~ubuntu24.04.1, release build

Hitting an issue where opening/saving a schematic file would not save the 'hide' text effect in symbols.
This is not an issue for symbol libraries.

## Schematic File version according to the top of the offending schematic file

```
(kicad_sch
	(version 20231120)
	(generator "eeschema")
	(generator_version "8.0")
```

## Minimum Test Script

```python
#!/usr/bin/env python3
import kiutils
from kiutils.symbol import SymbolLib
from kiutils.schematic import Schematic

import os
from os import path

import argparse
from pathlib import Path

# Arg Parsing
parser=argparse.ArgumentParser(description="Open Then Save and Close To Test Bug Replication")
parser.add_argument("src", type=Path)
p = parser.parse_args()
src_file = p.src.resolve()
base_path = path.dirname(path.realpath(__file__))

# Offending Lines
kicad_container = Schematic().from_file(src_file)
kicad_container.to_file(src_file)
```

brian-geicp, 2024-09-18:
Will create a minimal example schematic file later, but I can observe that in all symbol entities with property has it's effects lose both the styling like `(italic yes)` and `(hide yes)` after being opened/saved. Text position and font size is preserved.

mofosyne, 2024-09-20:
Attached a minimal example here now that replicates the problem

This is a minimal example of a KiCAD project where:
* Normal.kicad_sch : Unmodified control output 
* Test.kicad_sch : Processed by KiUtil

My test enviroment is 

```
$ kicad-cli --version
8.0.5
```

To run the test you can run `test.sh` which will run a python script to copy Normal to the Test file and then read/save Test.kicad_sch to check if it ends up corrupted.

I did find that it ended up corrupted so I created `sexp_linter.py` to at least harmonized the two document for easier comparison of what was corrupted between the control file and the test file.

[kiutilsbugtest.zip](https://github.com/user-attachments/files/17070298/kiutilsbugtest.zip)

mofosyne, 2024-09-20:
This diff shows that `hide` gets converted to `(hide yes)` when kicad upgrades to version 8:

[image]

This next diff screenshot shows the diff between a control file and a kiutil processed file

[image]

KiUtils is stripping out `(hide yes)` and removing quotes in `(uuid "97d75f03-218f-4681-9b81-727e962fa18e")` and turning it into `(uuid 97d75f03-218f-4681-9b81-727e962fa18e)` 

There may be more changes... but you can check the zipped file above to get a more indepth look on the issue

mofosyne, 2024-09-20:
In case anyone needs it, here is the gist for the s-exp linter/formatter that tries to match as closely as possible to kicadv8 formatting style https://gist.github.com/mofosyne/93ad8d64fb40ef80f6af3b09dc755414 to make kicad tool easier to develop

kevswims, 2024-10-14:
I believe at least the parsing issue with the hide field is caused by the check on this line: https://github.com/mvnmgrx/kiutils/blob/master/src/kiutils/items/common.py#L482 since it is looking for it to just be the `hide` field, and not the `(hide yes)`.

mofosyne, 2024-10-14:
Hmmm... I do wonder how kutils deal with version differences between file format version

T0jan, 2024-10-14:
Ohhh I was wondering why our libraries got broken all the time after I locally fixed the same issue with the kicad-library-utils, now it makes sense.  The fix of the IO format for them is also pending for quite a while now:

https://gitlab.com/kicad/libraries/kicad-library-utils/-/merge_requests/533